ZeroPhone/ZeroPhone-PCBs #3: EOMA specifications

dumblob, 2017-03-01:
I came across the FAQ on https://hackaday.io/project/19035-zerophone-a-raspberry-pi-smartphone/log/51834-explaining-choices-behind-the-project and wanted to mention, that there is a huge, more than 6 years running effort to:

1. Create EOMA open-source specifications of **interfaces** for HW (not the HW itself), which build upon open-source and libre concepts, eco-consciousness, sustainability over many years, and affordability.
1. Run a world-wide certification authority certifying compliance of devices with EOMA specifications.
1. Boost spreading of EOMA specifications by manufacturing some HW made according to these interfaces specifications (see e.g. https://www.crowdsupply.com/eoma68/micro-desktop ).

For more information visit http://elinux.org/Embedded_Open_Modular_Architecture and http://rhombus-tech.net/ (it's messy, but it's getting better; so far use Ctrl+F on http://rhombus-tech.net/sitemap/ ). Except for desktop PC, notebook and tablet, there is also a [phone](http://rhombus-tech.net/community_ideas/hybrid_phone/ ). These links account for about 40% of information. The whole rest is scattered in the [mailing list archives]http://lists.phcomp.co.uk/mailman/listinfo/arm-netbook ).

Some type of cooperation for the future ZeroPhone2 might be advantageous.

CRImier, 2017-03-04:
Hi! I've been following EOMA project (also, the EOMA68 implementation on CrowdSupply in particular) for some time. I like this concept, hope to get a EOMA68 laptop eventually and I'm considering this standard as one of those I could potentially use. ZeroPhone 2 isn't coming this year, but I'm already collecting information, so thank you for this =)

dumblob, 2017-03-04:
Wow, great news! I'm looking forward to cooperation and hope to hear from you sometimes at the EOMA mailing list.

(feel free to close this issue once there are more issues and things would get messy in the tracker; I'm leaving it open for other people to get a glimpse about a possible future)

CRImier, 2017-03-04:
I hope I'll join and participate in EOMA mailing list, eventually =) EOMA initiative is something that really needs to be worked on, and I'd be more than happy to be a part of it. 

Hopefully,  people will continue throwing ideas and links at me - Google doesn't always help =D