ZeroPhone/ZeroPhone-PCBs #10: Powering the host USB port from charger voltage while ZeroPhone is charging

CRImier, 2017-05-14:
The USB step-up board is one of the only things that's powered from the battery even when the phone is charging (other things are GSM modem, IR expansion header and the MCP23017 expander). The problem is that, when there's a charger plugged in the USB port, IMO it makes sense to feed the charger input directly into the USB port. It could be done in different ways - a SPDT switch, a FET switch or something else. Waiting for comments and pull requests =)

davedarko, 2018-02-16:
You could use a power muxer like the TPS2113A on the 5V line. Whenever it's charging, the 5V would come from the same source the charger get's. If there are no 5V the chip would grab the power from the booster. So while charging there wouldn't be much current drawn. But the chip is a bit pricey. Tricky situation as it seems you use 5V, 3V3 and 4V2 simultaneously from different sources.

CRImier, 2018-02-17:
There are all kinds of integrated components to do this kind of thing - however, I need to make sure that everything in the BOM is easy to source and solder. Thus, some specialised ICs aren't really suitable, and I build stuff from what I have in the BOM + things that are easy to source on places like eBay. Today, I added a 74HC125 and a LM339, for example - these are all popular parts, they're easy to get from local sellers, Digikey and eBay shops alike =) I do have a P-FET in the BOM (IRLML6401), as well as BC847's, SS14, 10K resistors and 100nF caps.

CRImier, 2018-02-17:
Also, 3V3 and 4V2 don't even come into play in this situation - or am I missing something?

davedarko, 2018-02-18:
There's a polulu module with that chip, but that doesn't get cheaper :( I don't think I fully understood the problem then. And it's late again. But I think power muxing can help you out. Or take a look at the diode solution that some adafruit boards feature. What about a mosfet, that turns itselve on? I think I saw something like that as a simple protection circuit.

CRImier, 2018-04-25:
So, what I added: when the 5V step-up DC-DC is off, there's a circuit (2 FETs back-to-back) that connects the charger input and the 5V output. As soon as the DC-DC is turned on, the charger input and the 5V output are no longer connected. This way, when the charger is connected, the USB port can be powered by the charger - you just need to make sure the DC-DC is turned off. As a side effect, this also allows charging the ZeroPhone using a USB male-to-male cable (which will likely be necessary for OTG anyway).

closed 2018-04-25 by CRImier