ZeroPhone/ZeroPhone-PCBs #12: Lack of 3D models for many components

CRImier, 2017-05-14:
KiCad provides board 3D view/export functionality, which would be very useful for creating ZeroPhone cases. However, for many components used by ZeroPhone there just aren't suitable 3D models. What's necessary is to find/develop simple to-scale 3D models for these components and link them to the PCB so that they appear in the 3D viewer, and can then be exported by the KiCad VRML export option for case building.

CRImier, 2017-05-14:
Addition - it's very important that 3D model source files are present, due to the fact that KiCad will be moving away from VRML in the future.

CRImier, 2017-05-17:
Components that are the most necessary for case creation:

Front board:
* Audio jack: JACK1 [(part number is FC68125)](https://crimier.wordpress.com/2017/05/16/raspberry-pi-audio-jack-part-number/)
* Microphone: W3 (currently has a terminal block 3D model, which is kinda unacceptable =) Can basically be a cylinder, [dimensions can be taken from this page](http://item.taobao.com/item.htm?id=17943296350))
* OLED screen: P2/P7 (any of those)
* ESP-12: U2 (can be taken [from this repository](https://github.com/jdunmire/kicad-ESP8266), license seems GPLv3-compatible)

Back board:
* Expansion headers: P1, P2, P3, P7 (2.54 THT angled headers)
* RGB LED: D3 (5050 case)
* SIM800 modem breakout: U2 (purely for the SIM card slot)
* TP4056 breakout: U4 (Chinese breakout board)
* 5V boost DC-DC breakout: U6 (Chinese breakout board)

Keypad board:

* 4-pin switches: UP1, DOWN1, LEFT1, RIGHT1, OK1 (SKQG switches, though with a higher pin part)
* 2-pin switches: N0-N9, N*1, N#1, C1-C6, GREEN1, RED1 (very popular and can be sold under many different codes, [photo here](http://hz00.i.aliimg.com/img/pb/795/696/038/1038696795_015.jpg))
* 4-pin/2-pin side switches: VOL_UP1, VOL_DOWN1, OPTION1, PHOTO1, PHOTO2 (EVQP7C switches )

dwaq, 2017-05-17:
In regards to ESP-12, are you using the 22 pin version or the 20 pin version? The 20 pin version uses pins 2-21.

CRImier, 2017-05-17:
@dwaq AFAIK the 3D model doesn't use the pins for model alignment on the board, instead, it uses the grid reference point, so there should be no problems =) 

If the schematic didn't make it clear (I'll re-check), I'm actually using the 22-pin model of ESP-12, but some pins are removed from the ESP-12 footprint, since they're not used anyway and there's more space for routing traces and antenna keepout polygon. If the 3D model has pins that just won't be on the pads, I imagine it's not going to be an issue - this is what it looks like in reality, on a real ZeroPhone =)

CRImier, 2017-05-18:
Found the SKQG button 3D rendering [here](https://grabcad.com/library/momentary-pushbutton-5-2mm-surface-mount-smd-1)

dwaq, 2017-05-18:
I'm working on the Microphone: W3 now. What are the dimensions? I don't see details on that page and can't tell if 6mm and 5mm is height and radius or what. I can't even make sense out of it while looking at the footprint on the PCB.

PS, you can use the task list markdown for the models remaining and add a check when they're done instead of crossing out. But that's personal preference I guess.

CRImier, 2017-05-18:
Found the 2-pin switch 3D model [here](http://smisioto.no-ip.org/elettronica/kicad/kicad-en.htm)
[image]
2-pin SMD button on the left to the SKQG button

CRImier, 2017-05-18:
Task list - done, thank you for suggestion!

dwaq, 2017-05-18:
As for the OLED display, I found [Monochrome 1.3" 128x64 OLED graphic display on Adafruit](https://www.adafruit.com/product/938) which appears to be compatible with your design, but not the exact model that you're using. It provides dimensions but not a 3D model so I'd have to design it. Should I use these dimensions or can you find the dimensions of the display you're using? For case design, I assume screen dimensions are going to be very important, so this model needs to be correct.

CRImier, 2017-05-18:
@dwaq - you're right about screen dimensions being very important. I can't measure my screen right now, and can't check if the dimensions are right, either, but I'll measure the screens I'm using and post the dimensions today or tomorrow.

CRImier, 2017-05-18:
@dwaq Just measured the microphone, it's 5mm in width and 6mm in height.

dwaq, 2017-05-18:
I tried modeling those dimensions today and it didn't look right. Can you provide a photo of the part soldered to the board so I can see how it's supposed to look? Thanks.

CRImier, 2017-05-18:
@dwaq Here it is:
[image]

CRImier, 2017-05-19:
Pushed new changes - changed the crystal footprint to a reasonable one and removed pin headers that aren't actually soldered on.

dwaq, 2017-05-19:
Here's a capture of the microphone modeled at 5mm width, 6mm height. As you can see, it doesn't fill out the terminal block silkscreen like the component in your photo does. It also barely covers the holes.
[image]
I'll make a pull request regardless and you can either let me know what to change, or modify it yourself.

CRImier, 2017-05-19:
Hi! It should be OK, mostly it's the center position and height of microphone that matters . I'll check its height once I'll think of a way to do that. As for the possible modifications - it would be good to have a thin black layer on the top, just like it is with microphones I'm using - but nothing more, you could just paint the top black.

CRImier, 2017-05-20:
@dwaq - sorry for the screen dimension delay. Screen dimensions added in #24 (so that this issue isn't picture-heavy).

CRImier, 2017-05-20:
Added #25 for TP4056 charger breakout board

CRImier, 2017-05-20:
Fixed pin headers on back board, chaned pushed to repo

dwaq, 2017-05-24:
For RGB LED: D3 (5050 package), are [these dimensions correct for the part you're using?](https://www.adafruit.com/product/619) Especially the location of the polarity mark? I found a few 3D models of 5050 LEDs but they don't match these dimensions. If I have to model it myself, it's just going to be a cube -- I can't make it as detailed as the 3D models out there.

CRImier, 2017-05-24:
@dwaq Yes, that's the correct type. As for the dimensions - could this just be a scaling issue?

dwaq, 2017-05-24:
[This one has the wrong dimensions (CREE must do it differently)](https://grabcad.com/library/5050-plcc6-led-1)
[This one has the polarity mark on the wrong location](https://grabcad.com/library/led-smd-5050-1)
[I found this one and converted the file format and it appears ok, but I need to get permission from him to use it](https://grabcad.com/library/ws2812b-rgb-led-adafruit-neopixel-1)

CRImier, 2017-05-24:
The first one: yeah, if it could be scaled, it would be very cool.
The third one is OK too - it's a 4-pin one (instead of 6 pins), but it's not too much of a problem since this model is meant to be a guide for where to make a cutout/light path in the case, not to be 100% pin-perfect.

dwaq, 2017-05-26:
All that's left is the modules.. These are too complicated to create using my current design flow. I don't know the "correct" way to make these. Create them in KiCad and export the 3D model? Create them from the ground up in some 3D cad system? If you know of a tutorial on how this is typically done, I might be able to try it. Otherwise, I'm going to leave these items for someone else to tackle.

CRImier, 2017-05-28:
You know, making the parts as KiCad PCBs and them exporting them does make a lot of sense, I could easily do that. For example, the current GSM module is basically a PCB with a GSM module, SIM slot and pin headers. I could easily model that, and then it could be exported to VRML and imported to another board. It's still going to need some simpler blocks, like the SIM holder, the GSM modem itself (basically a flat square block) and an inductor - would you be interested in drawing those from dimensions? That'd finish the 3D modelling stage, so we could start designing cases =)

CRImier, 2017-05-28:
Also, @dwaq - could you find a good 3D model of a Pi Zero? I think that's more or less the only external part we need to put it all together.

dwaq, 2017-05-30:
As far as the module comment.. Yeah, I think I can help make some components. It might be nice to make them in a separate repo so that others can more easily find them and use them and we can just import the product of that work into this model.

CRImier, 2017-06-09:
https://github.com/ZeroPhone/BreakoutPCBMockups - done =)