ZeroPhone/ZeroPhone-PCBs #15: back_prototype.sch old schematic component libraries

dwaq, 2017-05-17:
When I try to open [back_prototype.sch](https://github.com/ZeroPhone/ZeroPhone-PCBs/blob/master/back_pcb/back_prototype.sch), I see this helper menu. I'm not sure which part is correct, so it's probably best if you correct this issue.

[image]

CRImier, 2017-05-17:
Hi! I updated my KiCad to 4.0.6, and then pressed OK, and it seems to have saved the symbols instead of replacing them. Same was when I opened the front board. You can just press OK on these windows, and the issue will be solved on the next board file update, whoever of us does it first =)

CRImier, 2017-05-19:
Changes committed and pushed to the repo, thank you!

closed 2017-05-19 by CRImier