ZeroPhone/ZeroPhone-PCBs #33: JST connector for battery: P8 on back board

dwaq, 2017-05-25:
Making a new issue outside of #12 because this might get photo heavy again

I found [this 3D model](https://grabcad.com/library/2-pin-jst-connector-right-angle-1) and I was attempting to add it to the board but it doesn't seem to match with the footprint.

I have it aligned in the X (x is up/down, y is left/right in this photo) direction, but need details on the Y direction. First off, does this appear to be the correct size? It's smaller than the outline in the X direction. In the Y direction, where should it align to? The edge of the PCB? If so, what set of holes do the pins fit? The ones nearest the PCB edge or the ones that are set further away? Some photos of the actual board might help me.
[image]

CRImier, 2017-05-25:
Photos: will do. Actually, it's my fault - I haven't really thought about through-hole connectors, mostly because I didn't have any at the time I ws drawing this connectir footprint. This kind of connector you have the 3D model for is smaller in X because it is smaller IRL - it's a different kind of connector, however, it's actually more popular and I should make it fully compatible. 

I think you can just shift it inside the board for now, align with those holes that are deeper inside the board - the battery connector placement is awkward for now, shifting it in the board would make the connector stick out less. Also, if I won't be able to source SMD connectors like the one I'm using, next week when I'll be actually finishing assembly of ZeroPhone boards that arrived today, I'll certainly have to source some through-hole ones (like those that the model you have is for) and use them instead - and that'll be the way I'll use them.

Will send photos later today, now have to get to my workshop =)

CRImier, 2017-05-25:
Made the photos;

[image]
This is how the SMD connector looks right now. It's awkward because the plug with wires sticks out more now and is therefore more at risk of breaking, and I'll consider moving this connector further into the board - but not now.

I got one of the THT connectors, soldered them to the boards I have and this is how it looks:
[image]
[image]

IMO it's more reasonable, and given that I can't seem to actually source those SMD connectors here now, it seems like what I'll go with.

dwaq, 2017-05-25:
Straight down from the top - looks like it's close to the same size as yours:
[image]
Now here you can tell it's off a little because it's overlapping the red pads (where I circled) and yours aren't.
[image]
On the back you can see that the pins are a little too big for the hole.
[image]

This could possibly be a scaling issue, but I can't think of a way to know if it's scaled correctly or not. I scaled this model the same as all the others I've created. Should I move forward with adding this part as-is?

CRImier, 2017-05-25:
I believe the scaling doesn't matter as much here (especially as the possible error is so small), the main thing is being able to position an opening for the connector, so we can leave scaling as it is now =)

closed 2017-05-25 by CRImier