ZeroPhone/ZeroPhone-PCBs #57: Changing through-hole pads to oval ones

CRImier, 2018-02-17:
In order to make ZeroPhone more reworkable, we need to make sure that pads stay on the board. @dwaq advice is:

```
In regards to the pad delamination, oval pads are typically stronger because there's more surface area, yet you can retain the spacing between the pins.
```

So, some of the through-hole pads should be made oval. I think it makes sense to do it for 
- [ ] GSM modem
- [ ] front-to-keypad board connections
- [ ] JST connector
- [ ] USB port
- [ ] Microphone

Possible but problematic places:
- [ ] 40-pin Pi Zero headers (there are traces between them)
- [ ] Display header (8-pin and 7-pin holes are too close)
- [ ] Audio jack footprint (the routing around it is cramped)

Places where holes can't/shouldn't be 
- Both 6-pin headers - there are traces going near them
- Expansion headers - not meant to be reworked (little use in reworking them), and not enough place

dwaq, 2018-02-19:
I'll start work on this over at https://github.com/dwaq/ZeroPhone-PCBs/tree/oval-pads

CRImier, 2018-02-19:
Great, thank you! I just had to stop working on it, as I'm working on my side job. If you need any advice, materials (like photos) or something else, please let me know.

dwaq, 2018-02-19:
I made PR #58 which makes oval pads on the 5x parts you mentioned "makes sense". The pads are changed and the new footprints are imported into the boards (front and back pcbs were modified). Because the pads are different sizes now, they interact with the VCC and GND pours and those will need to be modified - I did not do this. The traces to these may not need to be modified, but they should be checked. 

As a general rule I made the pads 1.33x as big on one side as the other (this is what KiCad uses on the default library's oval pads). It's fairly simple to modify the footprints in a text editor if you think a different size would be better.

The microphone was a stock terminal block, so I made a new part from that and modified it to have oval pads.

dwaq, 2018-02-19:
[image]

As for the RPi header, you can make offset oval pads as seen in this screenshot. You can make it so they only point to the outside and you can still run a signal between the rows. There's some parts that will need to be moved if the footprint is modified in this way, so I did not attempt it. Another option is to only increase the size of the pads that have room to do so. At least the design would be improved by as many pads as you can modify.

CRImier, 2018-02-19:
Thank you! Could you also adjust the keypad board? (as it has the front-to-keypad connections footprint in it, too)

dwaq, 2018-02-19:
Oh ok, I didn't think of that. Will do

dwaq, 2018-02-19:
The keypad PCB now has the new footprint. I also had to regenerate and import the netlist to fix some DRC errors.  I think it was out of date.
The netlist in the `gitignore` so this board doesn't have one in the repo.
The front PCB has a netlist in the repo but I think it's also out of date because the PCB has DRC errors.  It's outside the scope of this issue but I wanted to alert you of it.

CRImier, 2018-02-19:
Thank you! I'll have to make sure I check all boards for DRC stuff, then - I have a habit of doing this, but now it seems I've missed the keypad board.

CRImier, 2018-02-22:
Checked it from my side, adjusted where necessary. @dwaq thank you for the help!

Thoughts:
- Audio header could indeed use oval holes, just in case (I never ever had to rework one, but I imagine it could become necessary in some case)
- Display header will definitely need oval holes (it's given me a lot of grief with the Gamma batch), however, care should be taken to preserve 7-pin display compatibility
- I'm not sure Pi Zero header could use oval holes - will need to think about it. I had some breakage while reworking a Pi Zero, but it's a tricky process by itself and people won't use it often.

dwaq, 2018-02-22:
If this issue is still open in a week, I can work on those 2 headers. Won't have time until then.

dwaq, 2018-03-13:
This is still on my radar, just got busy and haven't been able to work on it yet. I'll comment when I get a chance to start on it, but feel free to take it on yourself if you need it done before I get to it.

CRImier, 2018-03-16:
No problems, did it now =) Changed front board headers to use 1.524x1.9 oval pads - the rework there shouldn't be too hard anyway. Changed back board headers to 1.524x2 - I will see if rework is better, maybe I will need to increase the pins further.

CRImier, 2018-03-16:
Will try to improve the display headers now - other than that, this is completed! Thank you, @dwaq , for your help on this =)

closed 2018-03-16 by CRImier