ZeroPhone/ZeroPhone-PCBs #72: Conversion to KiCAD5

eeyrjmr, 2018-09-22:
Present project appears to be using KiCAD4 library structure. KiCAD5 changed this. Automatic migration not complete.

Request roadmap for toolchain upgrade

CRImier, 2018-10-04:
Likely when I'll finish my current ZeroPhone mod PCBs that I'm designing - after next week, then I will need to look into making cases, and could switch files to KiCad5 while we're at it.

eeyrjmr, 2019-01-16:
That is odd.. KiCAD presents a library mapping dialog

> This schematic currently uses the project symbol library list look up method for loading library symbols. KiCad will attempt to map the existing symbols to use the new symbol library table. Remapping will change some project files and schematics may not be compatible with older versions of KiCad. All files that are changed will be backed up to the “remap_backup” folder in the project folder should you need to revert any changes. If you choose to skip this step, you will be responsible for manually remapping the symbols.

and equally when git status is executed:

git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   front_pcb/front_prototype-cache.lib
	modified:   front_pcb/front_prototype-rescue.lib
	modified:   front_pcb/front_prototype.pro
	modified:   front_pcb/front_prototype.sch

CRImier, 2019-01-16:
Ah, I see, I misunderstood what you meant. Yeah, I had these changes too, will check the extent and whether I made the conversion fully, then push into a branch - or maybe even into master (with a disclaimer that files are no longer backwards compatible with 4 - that's true, right?)

eeyrjmr, 2019-01-26:
yup that is true.  
To be fair when I raised the bug it wasn't just project conversion, there were alot of parts missing (from the schematic and 3D symbols for the view).

when you responded I did a fresh pull and equally was on KiCAD-5.0.1  and the main issues were resolved (probably gentoo ebuild and library layout...) but the project conversion is still valid

CRImier, 2019-05-13:
Pushed the KiCad5-updated schematic files - I didn't push the PCB files yet, I need to bring my 3D model work results into an upload-able state.

closed 2023-01-09 by eeyrjmr